gmag11/QuickESPNow #1: QuickDebug dependency

rnapoli, 2022-06-14:
I'm using PlatformIO and installed the QuickEspNow Library for the first time, when compiling it cannot find the header <QuickDebug.h>, I had to manually install the Library dependency QuickDebug in PlatformIO.
After installing this dependency everything worked fine.

Target used: ESP32
Library: QuickEspNow 0.6.0

Thanks a lot for this library, Germán !!!

Compilation messages:
********************************************************************
********************************************************************
********************************************************************
In file included from .pio/libdeps/nodemcu-32s/QuickEspNow/src/QuickEspNow.h:5,
                 from .pio/libdeps/nodemcu-32s/QuickEspNow/src/QuickEspNow_esp8266.cpp:1:
.pio/libdeps/nodemcu-32s/QuickEspNow/src/QuickEspNow_esp32.h:15:10: fatal error: QuickDebug.h: No such file or directory

********************************************************************
* Looking for QuickDebug.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:QuickDebug.h"
* Web  > https://registry.platformio.org/search?q=header:QuickDebug.h
*
********************************************************************

 #include <QuickDebug.h>
          ^~~~~~~~~~~~~~
compilation terminated.

gmag11, 2022-06-23:
Hi. If using PlatformIO you should edit your platformio.ini to register this library as stated on PIO Registry: https://registry.platformio.org/libraries/gmag11/QuickDebug/installation

I hope it helps

closed 2022-06-23 by gmag11

gmag11, 2024-02-12:
With latest release (https://github.com/gmag11/QuickESPNow/releases/tag/0.7.0) QuickDebug is not required if you don't need to get debug messages

blazoncek, 2024-02-19:
It appears that 0.8.0 reintroduced QuickDebug dependency...

```
...
Removing unused dependencies...
Library Manager: Removing QuickEspNow @ 0.6.2+sha.b357ffa
Library Manager: QuickEspNow@0.6.2+sha.b357ffa has been removed!
Library Manager: Installing gmag11/QuickESPNow @ ~0.8.0
Unpacking  [####################################]  100%
Library Manager: QuickEspNow@0.8.0 has been installed!
Library Manager: Resolving dependencies...
Library Manager: Installing gmag11/QuickDebug @ 0.7.0
Unpacking  [####################################]  100%
Library Manager: QuickDebug@0.7.0 has been installed!
Verbose mode can be enabled via `-v, --verbose` option
...
```