gmag11/QuickESPNow #5: QuickDebug.h: No such file or directory

homoaroma, 2023-01-03:
Arduino\libraries\QuickESPNow-main\src/QuickEspNow_esp32.h:15:24: fatal error: QuickDebug.h: No such file or directory
compilation terminated.

homoaroma, 2023-01-03:
Sorry, I found out that I need to install another library.

blazoncek, 2023-08-19:
I would like this to be optional as I already use other debugging environment.

yanxke, 2024-01-04:
Should add a dependency so the library is automatically installed.

gmag11, 2024-01-07:
Hi, actually it is added as a dependency but it does not get installed for some people.
I use it for development without having to use print but I thought I could be useful for others.

But many people get confused with that message and maybe it is better to remove it.

What do you think?

blazoncek, 2024-01-07:
It is useful (I intend to integrate it into our project but will be major task) it is just unnecessarily included here when used as a library in other projects.

yanxke, 2024-01-07:
I tried to install it manually but it still doesn't compile. I think that library has other dependencies and so I ended up making dummy definitions for the macros and a skeleton .h file. It's the only way I could get this library to compile. I'm using PlatformIO FWIW.

gmag11, 2024-02-12:
With latest release QuickDebug is not required if you don't need to get debug messages

closed 2024-02-12 by gmag11