gmag11/QuickESPNow #8: Soft WDT reset

blazoncek, 2023-10-09:
First off, thank you for a very versatile library. It makes handling ESP-NOW very easy.

Unfortunately I am seeing "Soft WDT reset" on my ESP8266 installations when sending packets while connected to WiFi network.

My initialisation looks like this:
```c++
  if (enableESPNow) {
    quickEspNow.onDataRcvd(espNowReceiveCB);
    bool espNowOK;
    if (apActive) {
      DEBUG_PRINTLN(F("ESP-NOW initing in AP mode."));
      #ifdef ESP32
      quickEspNow.setWiFiBandwidth(WIFI_IF_AP, WIFI_BW_HT20); // Only needed for ESP32 in case you need coexistence with ESP8266 in the same network
      #endif //ESP32
      espNowOK = quickEspNow.begin(apChannel, WIFI_IF_AP);  // Same channel must be used for both AP and ESP-NOW
    } else {
      DEBUG_PRINTLN(F("ESP-NOW initing in STA mode."));
      espNowOK = quickEspNow.begin(); // Use no parameters to start ESP-NOW on same channel as WiFi, in STA mode
    }
    statusESPNow = espNowOK ? ESP_NOW_STATE_ON : ESP_NOW_STATE_ERROR;
  }
```
And for sending packets:
```c++
typedef struct PartialEspNowPacket {
  uint8_t magic;
  uint8_t packet;
  uint8_t segs;
  uint8_t data[247];
} partial_packet_t;

...

  if (enableESPNow && useESPNowSync && statusESPNow == ESP_NOW_STATE_ON) {
    partial_packet_t buffer = {'W', 0, (uint8_t)s, {0}};
    // send global data
    DEBUG_PRINTLN(F("ESP-NOW sending first packet.")); 
    memcpy(buffer.data, udpOut, 41);
    auto err = quickEspNow.send(ESPNOW_BROADCAST_ADDRESS, reinterpret_cast<const uint8_t*>(&buffer), 41+3);
    if (!err) {
      // send segment data
      buffer.packet++;
      size_t packetSize = 0;
      int32_t err = 0;
      for (size_t i = 0; i < s; i++) {
        memcpy(buffer.data + packetSize, &udpOut[41+i*UDP_SEG_SIZE], UDP_SEG_SIZE);
        packetSize += UDP_SEG_SIZE;
        if (packetSize + UDP_SEG_SIZE < sizeof(buffer.data)/sizeof(uint8_t)) continue;
        DEBUG_PRINTF("ESP-NOW sending packet: %d (%d)\n", (int)buffer.packet, packetSize+3);
        err = quickEspNow.send(ESPNOW_BROADCAST_ADDRESS, reinterpret_cast<const uint8_t*>(&buffer), packetSize+3);
        buffer.packet++;
        packetSize = 0;
        if (err) break;
      }
      if (!err && packetSize > 0) {
        DEBUG_PRINTF("ESP-NOW sending last packet: %d (%d)\n", (int)buffer.packet, packetSize+3);
        err = quickEspNow.send(ESPNOW_BROADCAST_ADDRESS, reinterpret_cast<const uint8_t*>(&buffer), packetSize+3);
      }
      if (err) {
        DEBUG_PRINTLN(F("ESP-NOW sending packet failed."));
      }
    }
  }
```
The snippet above deals with oversized packets and works well on ESP32 (also tested on ESP32-C3, but for some reason QuickEspNow will not work at all on ESP32-S2 but that is for another issue). The code above finishes/completes without issue on ESP8266 but a few seconds after it is executed soft watchdog triggers a reset.

I managed to dump crash log using exception decoder and it is attached below.
Apparently `espnowTxHandle()` stalls somewhere.

Do you have any idea what might be wrong?
[... 376 lines cut ...]
0x40100e30 in ets_post at ??:?
0x40106f49 in ets_timer_disarm at ??:?
0x4023b16f in handlePresets() at ??:?
0x402483b1 in WLED::loop() at ??:?
0x40296029 in cont_continue at cont.S.o:?
0x40264e88 in run_scheduled_recurrent_functions() at ??:?
0x4025247c in ESP8266WiFiSTAClass::localIP() at ??:?
0x4023f68c in NetworkClass::localIP() at ??:?
0x402705fc in StreamNull::~StreamNull() at ??:?
0x402705fc in StreamNull::~StreamNull() at ??:?
0x40106e09 in wdt_feed at ??:?
0x40267012 in yield at ??:?
0x40106eb6 in wdt_feed at ??:?
0x4010708d in ets_timer_arm_new at ??:?
0x40266f64 in __esp_suspend at ??:?
0x402670ad in __esp_delay at ??:?
0x40267123 in esp_try_delay(unsigned int, unsigned int, unsigned int) at ??:?
0x40267f88 in delay at ??:?
0x402483e1 in WLED::loop() at ??:?
0x402705fc in StreamNull::~StreamNull() at ??:?
0x40268e74 in uart_write at ??:?
0x40263670 in HardwareSerial::write(unsigned char const*, unsigned int) at ??:?

--------------- CUT HERE FOR EXCEPTION DECODER ---------------
```

blazoncek, 2024-02-12:
Hi. Thanks for the explanation. I did check `advancedespnow` several times when writing above code. I didn't think I needed actual callback on `dataSent` as I do not need any confirmation or other use for that.

Yes, we need to send (up to) 6 packets (about 1400 bytes) that comprise a full payload. Everything needs to work asynchronously without delaying main loop as it is used for realtime data (calculating LED effects).
I have assumed that QuickESPNOW handles such approach (and it seems it does well on ESP32).

BTW HardwareSerial might be a debugging trace.

gmag11, 2024-02-12:
QuickESPNOW works with a queue to send messages. It has a buffer for 3 messages. There is a process that consumes the queue and actually send messages through esp-now, but only if last packet is confirmed.

So, you can send up to 3 messages in a row without waiting, but if queue is full and you try to send a new message, then the older in the queue is discarded.

You can increase queue depth by tuning `ESPNOW_QUEUE_SIZE` into header files. It will consume about 300 bytes for every message.

I'm doing some changes in the library to allow synchronous sending, but, if your timing is critical, I would recommend to control it by yourself so that you may use full queue. With synchronous mode only one packet can be stored.

blazoncek, 2024-02-13:
Hi. I was able to (logically) trace where the problem is (it is twofold).

1) calling `delay()` function from within ISR should not happen (assuming `espnowTxTask_cb` is called from OS timer interrupt)
2) sending a few messages in quick succession may enter a loop that takes too much time for WDT

Let's say you `quickEspNow.send()` 4 messages (overflowing queue). First call to `espnowTxHandle()` will set `readyToSend` to `false` until sending of a single packet completes (in `tx_cb`) but there are still 2 meesages in queue so it will enter `while (!readyToSend) delay(1);` loop which may be enough to trigger WDT as `delay()` is not permitted in ISR.

What I would do is break out of `espnowTxHandle()` instead of entering `while (!readyToSend)` loop and wait for next timer invocation of `espnowTxTask_cb()`.

I still need to test my theory and may do a PR if it is correct and you think it worthwhile.

blazoncek, 2024-02-13:
Update: Seems to work well. No more WDT reset.

blazoncek, 2024-02-15:
One more thing.

When the ring buffer is full I would expect (and would be the correct way) for `quickEspNow.send()` to fail and not discard last message. That way it would be easy to handle such errors in the application.

closed 2024-02-19 by blazoncek