gmag11/QuickESPNow #9: Your RSSI issue

homonto, 2023-12-28:
Hi,

While browsing your library I found you have a problem with RSSI:

`quickEspNow.dataRcvd (rxMessage.srcAddress, rxMessage.payload, rxMessage.payload_len, rxMessage.rssi, broadcast); // rssi should be in dBm but it has added almost 100 dB. Do not know why`

this is because, you should get rssi from another function - another place: in the timeframe of receiving the message from the sender.
I did it this way:

first callback function definition:
```
void promiscuous_rx_cb(void *buf, wifi_promiscuous_pkt_type_t type)
{
  if (type != WIFI_PKT_MGMT)
    return;
  const wifi_promiscuous_pkt_t *ppkt = (wifi_promiscuous_pkt_t *)buf;
  Serial.println(ppkt->rx_ctrl.rssi);
}
```
then assignment:
```
  esp_wifi_set_promiscuous(0);
  esp_wifi_set_promiscuous_rx_cb(&promiscuous_rx_cb);
  esp_wifi_set_promiscuous(1);
```

This works for me 
Question is now: how to assign this rssi from callback to your function dataRcvd(...).
What I did is: another structure, update of the field "rssi" in promiscuous_rx_cb() and then using it according to the need.

When I tried your version the rssi is random, not consistent with the real rssi between the sending device and the receiving device

Hopefully it makes sense and is helpful - if you implement it in your library let me know please.

gmag11, 2024-02-12:
Thanks, I'll have a look to this.

gmag11, 2024-02-12:
I use `wifi_promiscuous_pkt_type_t` to get data from memory buffer without using promiscuous mode. I guess using that mode may affect WiFi connection if STA or AP mode is active. Have you tested this?