gmag11/QuickESPNow #14: If esp_now_send fails, readyToSend never becomes true again

brawner, 2024-08-20:
I believe the logic expects that `esp_now_send` will always call `tx_cb`, but I'm not sure that will happen in all error cases. That means `readyToSend` can stay false and block any more sends from happening.

In lines: https://github.com/gmag11/QuickESPNow/blob/main/src/QuickEspNow_esp32.cpp#L201-L216

andre-di, 2024-09-13:
I think I have the same issue happening.

I trying to use WiFi at the same time and that causes ESPNOW to give this error "E (12712) ESPNOW: Peer channel is not equal to the home channel, send fail!".

After this error I get 3 broadcasts out and then it seemingly does not get ready again.

tigramaan, 2025-05-04:
> I think I am having this issue right now actually.
> 
> I trying to use WiFi at the same time and that causes ESPNOW to give this error "E (12712) ESPNOW: Peer channel is not equal to the home channel, send fail!".
> 
> After this error I get 3 broadcasts out and then it seemingly does not get ready again.

You can use quickEspNow.begin(0); to follow WiFi channel

fabiotu, 2025-06-04:
> I believe the logic expects that `esp_now_send` will always call `tx_cb`, but I'm not sure that will happen in all error cases. That means `readyToSend` can stay false and block any more sends from [...]

I am having the exact problem: both in AP and STA mode, same channels for sender and receiver devices: the yield loop is infinite cause `waitingForConfirmation` never becomes true.
Stuck forever with log "--------- Waiting for send confirmation ". On WLED this causes the light to become inaccessible (WIfi hangs). 
Only solution for me for now is switching to asynchronous.

What could be the cause for send() to fail ?