gmag11/QuickESPNow #19: quickEspNow.stop() hangs, when quickEspNow has not been initialised

oe1wkl, 2025-03-17:
I need the espnow function only for certain tasks (selected from a  menu). When the task has ended, I do not know if espnow has been used (and hence has been  initialised), or not. If i try to do a quickEspNow.stop() to be on the safe side, this will hang forever if it has not been initialised. If I initialise a second time, I get an error (ESP_ERROR_CHECK failed: esp_err_t 0x3069 (ESP_ERR_ESPNOW_NOT_FOUND) at 0x4008d10c and the ESP reboots.
Is there a way to cleanly shutdown quickEspNow, that would not hang when it has not been used at all?

oe1wkl, 2025-03-18:
There seems to be an underlying issue: once you execute quickEspNow.stop() , an attempt to quickEspNow.begin (channel) will fail and lead to as ESP reset.

tigramaan, 2025-05-04:
I have a similar problem. quickEspNow.begin() after quickEspNow.stop() - causes backtrace and crashes the system